Xinyuan-LilyGO/T-Display-S3 #312: Issue with Two I2C Devices

CCS86, 2025-07-26:
Hello,

I am trying to run my T-Display-S3 with an SCD41 and an MS5611 sensor, both connected to I2C.

 I started with both connected to 17 & 18, but in trying to squash the issue, I switched to a second set of inputs:

 SCD41:  VCC -> 3.3V, GND -> GND, SDA -> 18, SCL -> 17
 MS5611: VCC -> 3.3V, GND -> GND, SDA -> 13, SCL -> 12

With both sensors connected, you can see that the MS5611 gets a good reading, but the SCD41 does not:

```
I2C buses initialized.
MS5611 Sensor Found.
SCD41 Sensor Found.
Initial pressure compensation set to: 991 mBar
Automatic Self-Calibration DISABLED.
SCD41 periodic measurement started.
Setup Complete. Starting main loop.
E (4990) i2c.master: I2C transaction unexpected nack detected
E (4991) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
E (4992) i2c.master: i2c_master_receive(1268): I2C transaction failed
Error reading measurement: Error code: 527
Invalid CO2 reading, skipping.
Invalid CO2 reading, skipping.
```

With the MS5611 power cut (but everything else still connected) I get good SCD41 readings:

```
I2C buses initialized.
E (1484) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
MS5611 Not Found! Continuing without pressure compensation.
SCD41 Sensor Found.
E (2528) i2c.master: I2C transaction timeout detected
E (2528) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
E (2529) i2c.master: i2c_master_multi_buffer_transmit(1214): I2C transaction failed
Automatic Self-Calibration DISABLED.
SCD41 periodic measurement started.
Setup Complete. Starting main loop.
E (4990) i2c.master: I2C transaction unexpected nack detected
E (4990) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
E (4992) i2c.master: i2c_master_receive(1268): I2C transaction failed
Error reading measurement: Error code: 527
CO2: 913 ppm, Temp: 30.9 C, Humidity: 40.5 %
CO2: 863 ppm, Temp: 30.6 C, Humidity: 41.5 %
```

Any ideas?

lewisxhe, 2025-07-26:
1. Run I2C device scan to check whether the I2C device address is obtained correctly
2. Check whether there is a pull-up resistor on the external I2C bus
3. Check whether the device address conflicts

CCS86, 2025-07-27:
Thank you for your suggestions!

I tried some additional troubleshooting, like withholding power from the MS5611 until the SCD41 was providing good readings and that helped me find the issue. Once the MS5611 came online, the CO2 readings started plummeting. This was because the pressure value was being passed to the SCD41 library in the wrong units, effectively compensating the value to 0, and making it look like invalide sensor data.

I'm just a hack, leaning heavily on LLM, but I got it working.

lewisxhe, 2025-07-28:
Looks like there is a conflict between two devices. Did you resolve it?

CCS86, 2025-07-29:
It was acting like a conflict, but it was just bad units on a passed compensation value.

lewisxhe, 2025-07-29:
All right