Xinyuan-LilyGO/T-Display-S3 #314: Setup Failed: ESP_ERR_NOT_SUPPORTED after updating to ESPHome 2025.7.5

naamah75, 2025-08-08:
After updating to ESPHome 2025.7.5 I got this error:

```
[14:02:46][I][app:149]: ESPHome version 2025.7.5 compiled on Aug  8 2025, 13:05:43
[14:02:46][E][esp32_camera:127]:   Setup Failed: ESP_ERR_NOT_SUPPORTED
[14:02:46][E][component:141]:   esp32_camera is marked FAILED: unspecified
[14:02:46][W][safe_mode:030]: Last reset occurred too quickly; will be invoked in 8 restarts
[14:02:46][E][esp32_camera_web_server:099]:   Setup Failed
[14:02:46][E][component:141]:   esp32_camera_web_server is marked FAILED: unspecified
```

no changes to ESPHome YAML file.

lewisxhe, 2025-08-09:
What board do you use? T-DisplayS3 + Camera?

lewisxhe, 2025-08-09:
I don't understand what has been changed in the latest version. It was normal in the old version. I don't have this board now. You can try uploading the camera sketch from the repository to see if the camera starts normally.

naamah75, 2025-08-11:
> to check if everything works do you mean [this ](https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/tree/master/firmware)firmware to load?

The hardware is fine, because it works well with the original firmware.

lewisxhe, 2025-08-11:
If the original firmware is OK, then the problem is on esphome's side. I can't help you solve this problem for now. You can roll back to the old version of esphome and use it.

naamah75, 2025-08-11:
The problem is that now you need to specify the I2C bus ID instead of the pin inside the esp32_camera component.

closed 2025-08-11 by naamah75

lewisxhe, 2025-08-12:
OK, so how did you solve it? Could you please post it for others who might run into this problem later?